petterreinholdtsen/battery-stats #1: suspend/resume hooks don't run in jessie

anarcat, 2016-03-15:
for some reason, suspend/resume hooks don't run here and the data doesn't get collected then.

i have /etc/pm stuff installed here:

```
$ dpkg -S /etc/pm
unattended-upgrades, toshset, pm-utils: /etc/pm
```

i am not sure they get fired when my laptop gets suspended/resumed, but then again, i am not sure how that happens in the first place. it *may* be systemd doing the suspend, but it could also be the laptop's BIOS as well: it's unclear. i know that suspend happens when i close the lid, but that's about it.

later:

http://askubuntu.com/questions/609695/does-systemd-read-etc-pm explain
that /etc/pm/ is ignored by systemd, and suggest to provide units in
/lib/systemd/system-sleep/ instead.

copy from https://github.com/petterreinholdtsen/battery-status/issues/8, see there for more comments

petterreinholdtsen, 2016-03-22:
When I test, the trigger runs on susped/resume, but not when I insert and remove the power plug.  I just commited a fix for the latter, by adding udeb rules to run the collector when the power status changes.

Are you sure the collector do not run when you suspend?

anarcat, 2016-03-23:
to be honest, i am not sure anymore - i just copied the issue from the other project.

petterreinholdtsen, 2016-03-23:
Right.  Then I suspect this issue is fixed, and the fix was included in version 0.5.1.

closed 2016-03-23 by petterreinholdtsen