petterreinholdtsen/battery-stats #2: Support multiple batteries in battery-status-graph

anarcat, 2016-03-15:
copy of https://github.com/petterreinholdtsen/battery-status/issues/9 - from @solarnz.

battery-status-collect will collect statistics for all batteries that are present on the system, however battery-status-graph assumes that there is only one battery in the log file.

It would be great if battery-status-graph was able to generate separate graphs for each different battery.

anarcat, 2016-03-15:
if you are curious, this may be possible / easier with the numpy/matplotlib stuff in the new python grapher i worked on in #10.

i am tearing my hair out on that stuff right now, so i'd gladly let someone else deal with it as well. :)

there are ways to produce multiple figures at once - it seems to me that seperate batteries should have separate plots, something like this:

[image]

withou the error bars of course...

ndechesne, 2016-03-15:
i am wondering why you said that battery-stats-collector would log information about all batteries, from what I am seeing it would only collect the info for 1 battery only. there is a for loop, but there is a break statement.

on my system with 2 batteries, the log only contain information about 1 battery, not 2. and yes, I agree it would be nice to see them all.

petterreinholdtsen, 2016-03-15:
Note, anarcat talked about battery-status-collect, which is a different program from battery-stats-collector.  It was the name of collect-csv when the package was named battery-status.  I'm in the process of adding two collectors to the package, one collecting the original format to /var/log/battery-stats, the other collecting in a new format to /var/log/battery-stats.csv.

anarcat, 2018-08-26:
so with the patches in #28  and #25, it's actually possible to graph multiple batteries at once. they just all end up in the same graph, which is kind of ugly and confusing, but works.

here's an example:

[image]

thanks to #25, it doesn't freak out when trying to guess battery expiry and still draws the graph:

```
$ ./src/battery-stats-graph --output ~/Downloads/batt.png
loading CSV file ['/var/log/battery-stats.csv'] with builtin CSV module
building data array
guessing expiry
could not guess battery expiry: On entry to DLASCL parameter number 4 had an illegal value
building graph
drawing to file <open file '/home/anarcat/Downloads/batt.png', mode 'w' at 0x7f2c247c2b70>
```

It's also possible (but hackish) to grep only for one battery identifier and generate a graph just for that:

[image]

And then it can guess the expiry correctly:

```
$ (head -1 /var/log/battery-stats.csv ; grep -a 45N1025 '/var/log/battery-stats.csv') | ./src/battery-stats-graph --output ~/Downloads/batt.png -
loading CSV file ['-'] with builtin CSV module
building data array
guessing expiry
this battery will reach end of life (5%) in 100 days, 4:26:53.646381, on 2018-12-04 20:55:54.157078
building graph
drawing to file <open file '/home/anarcat/Downloads/batt.png', mode 'w' at 0x7f9585983b70>
```

I'm happy this works, but I'm sad my battery is such crap!