petterreinholdtsen/battery-stats #4: old python graphing script broken

anarcat, 2016-03-24:
the python script i wrote is still in the package, but not the default and worse, it's actually broken now:

```
$ battery-log | ./src/battery-stats-graph
loading CSV file <open file '<stdin>', mode 'r' at 0x7f11f0fe10c0> with builtin CSV module
building data array
guessing expiry
Traceback (most recent call last):
  File "./src/battery-stats-graph", line 190, in <module>
    death = guess_expiry(data[full], data['timestamp'], zero)
  File "./src/battery-stats-graph", line 162, in guess_expiry
    fit = np.polyfit(data[full], data['timestamp'], 1)
  File "/usr/lib/python2.7/dist-packages/numpy/lib/polynomial.py", line 581, in polyfit
    c, resids, rank, s = lstsq(lhs, rhs, rcond)
  File "/usr/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 1867, in lstsq
    0, work, lwork, iwork, 0)
ValueError: On entry to DLASCL parameter number 4 had an illegal value
```

my guess is that it wasn't adapted to the new CSV structure, or not correctly. it fails at guessing the death of the battery, above...

petterreinholdtsen, 2016-03-24:
You are feeding it the wrong log file.  The battery-log script output the original log, which lack the information needed by battery-stats-graph.  Use the battery-graph with battery-log.

anarcat, 2016-03-25:
i think this was fixed in #5.

closed 2016-03-25 by anarcat