petterreinholdtsen/battery-stats #6: battery-stats-graph-flow fails to do anything

anarcat, 2016-03-24:
i get the following when running the flow program on my logfile:

```
[1239]anarcat@angela:battery-stats$ /usr/share/battery-stats/battery-stats-graph-flow
Argument "" isn't numeric in subtraction (-) at - line 17, <$fh> line 4.
Argument "" isn't numeric in subtraction (-) at - line 17, <$fh> line 4.
Argument "" isn't numeric in subtraction (-) at - line 17, <$fh> line 5.
Argument "" isn't numeric in subtraction (-) at - line 17, <$fh> line 5.
Argument "" isn't numeric in subtraction (-) at - line 17, <$fh> line 6.
Argument "" isn't numeric in subtraction (-) at - line 17, <$fh> line 7.
Argument "" isn't numeric in subtraction (-) at - line 17, <$fh> line 8.
Warning: empty y range [0:0], adjusting to [-1:1]
```

now, maybe it's because i have a screwed up CSV logfile, because I have used the procedure in #3 to convert my previous CSV files... you can reproduce the issue with the following .csv file:

```
timestamp,when,manufacturer,,technology,serial_number,status,cycle_count,energy_now,energy_full,energy_full_design,charge_now,charge_full,charge_full_design
1443139045,2015-09-24T19:57:25-0400,SANYO,,Li-ion,3016,Charging,0,,,,974000,6602000,6600000
1443139201,2015-09-24T20:00:01-0400,SANYO,,Li-ion,3016,Charging,0,,,,1078000,6602000,6600000
1443139621,2015-09-24T20:07:01-0400,SANYO,,Li-ion,3016,Charging,0,,,,1360000,6602000,6600000
1443139801,2015-09-24T20:10:01-0400,SANYO,,Li-ion,3016,Charging,0,,,,1480000,6602000,6600000
1443140401,2015-09-24T20:20:01-0400,SANYO,,Li-ion,3016,Charging,0,,,,1882000,6602000,6600000
1443141001,2015-09-24T20:30:01-0400,SANYO,,Li-ion,3016,Charging,0,,,,2283000,6602000,6600000
1443141601,2015-09-24T20:40:01-0400,SANYO,,Li-ion,3016,Charging,0,,,,2685000,6602000,6600000
```

not quite sure what is going on here...

petterreinholdtsen, 2016-03-24:
Hm, look like your battery is different from mine, as mine only have the energy* fields filled in.  Perhaps the script need to be changed to cope with this?

anarcat, 2016-03-25:
yeah, i guess? not sure - the python grapher has hooks for that, that you added, i think. :)