petterreinholdtsen/battery-stats #8: keep more than 5 weeks of logs

anarcat, 2016-03-24:
this is a rare case of when i think it is relevant to keep a larger amount of logs. the current configuration of the debian package only ships with 5 weeks of logs. that is not a long time to do proper analysis! for example, here's a graph i generated with the data I have so far, which is about 6 months:

[image]

it is barely usable, more samples would be great! this log file takes less than 1MB:

```
868K    /var/log/battery-stats.csv
```

and will take 10 times less that when compressed:

```
$ gzip -c /var/log/battery-stats.csv  | wc
    226    1280   71159
```

so i would suggest bumping log rotation to "monthly" at least, and make that at least a year. I have set it to 4 years here, but that may be a little extreme:

```
/var/log/battery-stats {
    rotate 208
    delaycompress
    missingok
    notifempty
    nocopytruncate
    nomail
    weekly
    postrotate
        if which invoke-rc.d >/dev/null 2>&1; then
          invoke-rc.d battery-stats restart > /dev/null
        else
          /etc/init.d/battery-stats restart > /dev/null
        fi
    endscript
}
```

thanks!

petterreinholdtsen, 2016-03-24:
My intention was to keep the old rotation rule for the old log file battery-stats, while not rotating the new log file battery-stats.csv at all.  Isn't this working?  The old log file do not really make sense to keep around, as it lack the details needed for long term analysis, while the new one can be used for that.

anarcat, 2016-03-25:
hmm... i thought this was about `/var/log/battery-stats.csv` - i misread it.

well, i liked the idea of rotated log files! :) but i am not sure rotating the old logfiles is a proper way of dealing with this - they should have been renamed.

btw, the logfiles were named `/var/log/hjemmenett-battery-status.log` in the previous version, not `/var/log/battery-stats` which may be what got me confused.

furthermore, i am not sure rotating CSV files makes sense, because the headers will be only in the last logfile, and eventually disappear... or are they re-added by the collection scripts?