petterreinholdtsen/battery-stats #23: `battery-stats-collector` should just issue usable lines on stdout and no warning.

b3, 2016-09-28:
In `battery-stats-collector` when `get_logline` is unable to determine a value for `ac_online` it echoes a message warning that "*No power supply found*". This line then goes to the log file before the status line is echoed.

That way the generated file becomes unusable since every two lines it contains non-data. On my host it looks likes something like :

    No power supply found
    1474886747 56.11 1 2016/09/26 12:45:47 -
    No power supply found
    1474886807 56.45 1 2016/09/26 12:46:47 -
    No power supply found
    1474886867 56.82 1 2016/09/26 12:47:47 -

The warning line should not be echoed, or at least not on stdout.

b3, 2016-09-28:
I created a branch on my fork which propose a solution to the issue : b3/battery-stats@7ea441d1295abbe83a80323f40a27732d6d52553

b3, 2016-09-28:
I do not think that the missing power supply should cause a cron email. In fact I am unsure that this message should be raised at all. Without power supply detection `battery-stats-collector` still work.

My idea for the proposition was that if the connected status (online or offline) can be detected it should be logged, but if it can not be detected that information shoud be logged too.

That was why I proposed 3 states.

I do not presuppose any use of that last information though.

petterreinholdtsen, 2016-09-28:
Right.  make sense to me.

b3, 2016-09-29:
Do you think my proposition should be integrated in your code? If yes I can just make a pull request of my demo branch.

petterreinholdtsen, 2016-09-30:
[Bruno BEAUFILS]
> Do you think my proposition should be integrated in your code? If yes
> I can just make a pull request of my demo branch.

Yes.  Please make a pull request.

-- 
Happy hacking
Petter Reinholdtsen