petterreinholdtsen/battery-stats #37: Save graph as image from command

jagjordi, 2023-01-07:
Is there any way to save the graph as an image? As a flag to the `battery-graph` command?

petterreinholdtsen, 2023-01-07:
[Jordi Altayó]
> Is there any way to save the graph as an image? As a flag to the
> `battery-graph` command?

I would use my desktop screen save option available via the PrtSc key
and grab a picture of the graph window.

-- 
Happy hacking
Petter Reinholdtsen

jagjordi, 2023-01-07:
I am aware its possible to print screen, but I want to include the picture as part of a script to display it in a yad popup window, so it needs to be as part of the gnuplot call

jagjordi, 2023-01-07:
I saw that in the script `battery-stats-graph-gnuplot` there is some way to save a PNG, is that also available in the `battery-graph` command?

petterreinholdtsen, 2023-01-07:
[Jordi Altayó]
> I saw that in the script `battery-stats-graph-gnuplot` there is some
> way to save a PNG, is that also available in the `battery-graph`
> command?

I do not plan to implement such feature, but would be happy to accept a
well written patch to support making pictures of the graph
non-interactively.

-- 
Happy hacking
Petter Reinholdtsen