petterreinholdtsen/battery-stats #39: Not enumerating power_supply folder correctly

Psykar, 2024-01-02:
`# battery-stats-collector[22270]: cat: /sys/class/power_supply//sys/class/power_supply/AC/online: No such file or directory`

Note the double slash. This was after a recent update from git.

I'm using the AUR git package here https://aur.archlinux.org/packages/battery-stats-git

`2d3675357de73023b29f3cb4fc9b882624f871a2` was working fine.

petterreinholdtsen, 2024-01-03:
[Louis des Landes]
> `# battery-stats-collector[22270]: cat:
> /sys/class/power_supply//sys/class/power_supply/AC/online: No such
> file or directory`

Ah, sorry about that.  My mistake.  Fixed now in master.

-- 
Happy hacking
Petter Reinholdtsen

Psykar, 2024-01-03:
That did it, thanks! Super quick :D

closed 2024-01-03 by Psykar