petterreinholdtsen/battery-stats #40: gnuplot >=6.x no longer has '-title' argument

Psykar, 2024-01-16:
As per title.

```
unrecognized option -title
line 0: Cannot load input from 'Battery Graph'
```

Or with bash -x
```
+ gnuplot -persist -title 'Battery Graph'
+ echo 'zero_time_err = sqrt((A/B**2*B_err)**2+(1.0/B*A_err)**2) }'
+ echo 'full_time = (100-A)/B'
+ echo 'g(x) = x<zero_time?(x>full_time?f(x):1/0):1/0'
+ echo 'set xrange [ 758684676 : 758713476 ] noreverse'
+ echo plot '"/tmp/tmp.tk7Rtpedcl" using ($1 - 946733400):($3==2?$2:1/0) smooth unique axis x1y1 title "Plugged in" with lines linecolor rgb "blue" linewidth 3, 	"" using ($1 - 946733400):($3==1?$2:1/0) smooth unique axis x1y1 title "Discharging" with lines linecolor rgb "red" linewidth 3, 	"" using ($1 - 946733400):($3==0?$2:1/0) smooth unique axis x1y1 title "Unknown" with lines linecolor rgb "gray" linewidth 3 \'
+ true
+ echo ', g(x -(1705388544-946733400) ) title (B<0?sprintf("slope= (%.2f +/- %.2f) %/h", B*3600, B_err*3600):"") lc rgb "black" lt 2 '
unrecognized option -title
line 0: Cannot load input from 'Battery Graph'
```

Unfortunately I can't interpret what is needed from gnuplots changelog or I'd do an MR.

daniilkk, 2024-02-11:
I don't know what is the correct portable fix but if you use qt output (like me) you can just set title like this

tgy, 2024-04-02:
Got the same error on the latest git version :-/

bonswouar, 2024-12-22:
> I don't know what is the correct portable fix but if you use qt output (like me) you can just set title like this

I guess the clean way would be to check if Qt is available, otherwise check for kitty/x11/etc
...Or we can just assume the user has Qt :roll_eyes: 
Actually, according to my package manager, on Fedora at least `libQt6` is a dependency of  `gnuplot`

Btw I've updated my fork with some other fixes for gnuplot 6.0 _(a cpanel warning & local time broken)_, I don't know if I should do a PR though